gregsaun/bear_extruder_and_x_axis #3: Suggestion - RJ4JP

albegouws, 2018-05-15:
Where you use Igus bearings would it be beneficial to have fine extrusions in the bearing spaces to ensure a better fit?  In the gaps where the bearing would have a clip fitting?

gregsaun, 2018-05-15:
I can make it a bit more tight but the way the part is printed (and the fact that customer's printer will have different calibration) will make impossible to have the perfect fit for Igus bearing. Igus needs an aluminum case with tight tolerance to be properly mounted.

albegouws, 2018-05-15:
Good point.  I’ll give a shot anyway and see how it turns out.  Are you planning any major commits soon?  I’ll try and print the parts this weekend for my Mk2.5 Bear 😉

gregsaun, 2018-05-15:
I will push new commits in few minutes ;-)

closed 2018-06-07 by gregsaun