gregsaun/bear_extruder_and_x_axis #4: Cable guide back a - difficult print

albegouws, 2018-05-18:
Printing the parts right now, the a part of the cable guide in the default orientation is a very difficult print to do successfully, the initial two or so layers are only time little marks on the bed, after a few layers it comes loose - depending on speed obviously. You can rotate and print at an angle but I think the one side could be flattened to print easier? 

Will let you know how the rest goes, MK3 and Mk2.5 running at full speed.

gregsaun, 2018-05-20:
Cable guide back a and b need to be printed with other parts because they are very small. However I think I can use another orientation, I'll give a try ;-)

albegouws, 2018-05-20:
I flipped it on the flat side and adjusted angle to 5 degrees.

gregsaun, 2018-05-26:
Hello, I did see that my parts was really not correctly aligned and I was thinking that you were asking another orientation. Sorry for the inconveniences!

I am going to fix it now and make it just a bit wider too

gregsaun, 2018-05-26:
Done! Thanks for the help

closed 2018-05-26 by gregsaun