gregsaun/bear_extruder_and_x_axis #6: Feedback - Build completed

albegouws, 2018-05-20:
Just completed the build.  I reset the firmware to defaults and went through the whole self check again.  Calibration completed fine.

Build notes - X axis belts are 100% parallel now.  Distance between top and bottom of belts only differ by less than 1mm on 4 spots the whole length of the axis. X axis tension adjustment works perfect.  

Inexperienced users should not attempt without proper instructions though. 

X - carriage install was ok - however with no cable ties the install was a tad difficult especially when routing cables due to the weight of the extruder motor. On that - cable routing improved over original.  Filament sensor works ok so far, adjusting the tension on the bondtech easier with one screw too.  Unloading of filament seems easier with less of an blob than I had on the old setup.  Anecdotal for now, will tell you if that changes. 

As a note I added this : https://www.thingiverse.com/thing:2904281.  It works ok-ish although the design can be improved to sit better in the v-slot and adjusting from the bottom is not really easy.  Can do with refinement, will see if I can improve on it when I can steal some time.

albegouws, 2018-06-01:
Hi - thought I'd ask here instead of opening a new issue.  Simple question - on the updates you note improved print quality and selective infill.  Would you mind explaining what that means ?  Not sure I get it.
Thanks!

zbrozek, 2018-07-03:
Just built one up. More feedback:

- The zip tie slots on the tail don't have an immediately obvious direction for retaining the upper harness vs lower harness.
- The x-carriage is super hard to install, especially during attaching the belts.
- Belt retention teeth are a bit tight. Maybe because I didn't print with linear advance.
- There doesn't appear to be any place for the nylon wire strain relief...?
- Would be nice to have a printed M10 thread in the sensor cover to accept a pneumatic fitting.
- Oh man the built in x-tensioner is so nice.
- Would be nice if you included a nozzle fan in the archive. I forgot to make one. Oops.
- Near the Rambo mount (on full Bear MK2.5) the harness tail gets *really* close to the case.

trevjonez, 2018-10-09:
Put this on my mk3 sans any bear frame. working great thus far. I didn't even add the x rod adjustment screws everything just felt great without. good work on the filament sensor cable port. my oem parts didn't retain the cable at all. yours fit like a glove. I assembled with a fresh hotend and your whole filament path appears perfect. I could see the .4 nozzle opening straight through the fully assembled extruder.

one thing that would make assembly a bit easier is if you could open up the gap on the bottom half of the x carriage back so that the inevitable on/off cycles are a bit easier to do. It works as is. but there is no wiggle to it. might be inexperience with the part but seems it could be 2mm or so wider without any strength issues.

when you rework the instructions make sure you put a warning on the x tensioner screw length. anything over 18mm will damage the part. I tried a 20mm and pushed in the wall a bit before I realized what was happening. part still working fine but something to look out for.

then on the x idler rod. I ordered a 3mm rod and cut it to length. but my pulleys were made based on the major diam of an m3 screw so really I needed like a 2.95mm rod. at that point I just said _**screw**_ it and grabbed my dremel and cut the head off a screw and used that. so far all is well. might be worth pointing out that a 3mm bore pulley on amazon might not actually accept a 3mm rod. more of a user experience thing than an issue with your design.

I installed this with the JLTX fixed z mounts at the same time. these x ends plus those z ends are the perfect setup for the stock mk3 frame. far more rigid, and the opening on the z makes install much easier.

next for me is to add the JLTX indirect sensor mod. looks like the hole pattern on the stock cover match yours so I am hoping his interposer is a bolt on fit. I might open up the fusion file and see if I can integrate his mod directly into your extruder body. lower part count would be a plus IMO.

also not sure if possible but it would be rad to replicate the adjustable pinda mount concept from the skelestruder.

all in all great work. this is more proof to me that scad workflow, while seeming cool is a nail in the foot of any decent designer/engineer. three production revisions from prusa and your alpha extruder is better IMO. key difference seems to be the design workflow making experimentation much simpler, but I speculate.

gregsaun, 2018-10-12:
@trevjonez Thanks you very much for this detailed feedback, very useful!

> one thing that would make assembly a bit easier is if you could open up the gap on the bottom half of the x carriage back so that the inevitable on/off cycles are a bit easier to do. It works as is. [...]

Are you talking about this opening in green?
[image]

> when you rework the instructions make sure you put a warning on the x tensioner screw length. anything over 18mm will damage the part. I tried a 20mm and pushed in the wall a bit before I realized what was happening. part still working fine but something to look out for.

It is true that 20mm and 18mm screws are very close and mistake can be easy. As my extruder will no use 20mm screws I will put a warning on the beginning of the assembly instruction

> then on the x idler rod. I ordered a 3mm rod and cut it to length. but my pulleys were made based on the major diam of an m3 screw so really I needed like a 2.95mm rod. at that point I just said [...]

I know this is difficult to find proper size. I have improved description, hopefully it is good now. I am also planning to sell a hardware kit for the extruder with everything included to help people finding the hardware.

> I installed this with the JLTX fixed z mounts at the same time. these x ends plus those z ends are the perfect setup for the stock mk3 frame. far more rigid, and the opening on the z makes install much easier.

JLTX did a remix of my mounts in fact :-D

> next for me is to add the JLTX indirect sensor mod. looks like the hole pattern on the stock cover match yours so I am hoping his interposer is a bolt on fit. I might open up the fusion file and see if I can integrate his mod directly into your extruder body. lower part count would be a plus IMO.

This is a very nice mod from JLTX! Please keep me in touch with this, I am very very interested. I will not do this on final version because I want to avoid firmware update. However, this could be added in optional parts for advanced users

> also not sure if possible but it would be rad to replicate the adjustable pinda mount concept from the skelestruder.

I do understand and the way JLTX did it is very clever. However, I prefer to keep it simple and light as it is today. You do not need huge precision here and you are not touching it very often (for most people almost never).

> all in all great work. this is more proof to me that scad workflow, while seeming cool is a nail in the foot of any decent designer/engineer. three production revisions from prusa and your alpha [...]

3D printing enable the possibilities to create very very complex parts and with OpenScad this can take a huge amount of time. Prusa is definitely not using the full power of OpenScad either and could make all this much more convenient and powerful. However, this is just the software, it does not really help you to things correctly neither tells you were you want to go. I am taking lot of time to watch how parts are integrated in "real life", not on the computer. I take the time to tear down assemblies, do measurements, print test parts and jigs... Once I see where I can make this better then I go on CAD. Once I have something to start, I print it and restart the "real life test". I try to keep my parts sometimes to show how much versions I am doing but it always takes too much place :-D

trevjonez, 2018-10-12:
> Are you talking about this opening in green?
> [image]

That is exactly the area that might use a bit more tolerance. so long as it doesn't harm function of course.

> This is a very nice mod from JLTX! Please keep me in touch with this, I am very very interested. I will not do this on final version because I want to avoid firmware update. However, this could be added in optional parts for advanced users

I think what I want to do before I try the interposer is add a setting to the prusa firmware so that the filament sensor direction can be swapped on the lcd and/or via gcode command as to avoid firmware flashing while playing with it. As I think more about it this week, I don't think I will try to mod the body of your design to be indirect. I fear the bearing accessibility and idler door interaction would be overly tedious.

matthew-humphrey, 2018-10-12:
@gregsaun why would the jltx indirect sensor mod require a firmware update? My understanding is he's just contacting the filament with a bearing and the sensor is reading the presence of a bearing rather than the filament (so the bearing is proxying the filament presence). This is just a mechanical change, no firmware change needed.

trevjonez, 2018-10-13:
> @gregsaun why would the jltx indirect sensor mod require a firmware update? My understanding is he's just contacting the filament with a bearing and the sensor is reading the presence of a bearing rather than the filament (so the bearing is proxying the filament presence). This is just a mechanical change, no firmware change needed.
> 
> EDIT: Maybe firmware update because with the interposer there is an additional length of filament from detection before the filament is truly out?

The filament moving down spins a bearing. the sensor observes the other side of the bearing which is then moving up. The sensor is an IR laser that reports velocity and direction. the firmware mod required is simply changing a += operation to a -= operation so that the rest of the system still observes downward filament motion as down, etc.

gregsaun, 2018-10-14:
@trevjonez perfect explanation trevjonez, thanks!

>That is exactly the area that might use a bit more tolerance. so long as it doesn't harm function of course.
Ok I'll check if I can do something, it might sound that it will not perfectly hold the bearing... I am personally moving the spiral wrap a bit on the back, not the most safe method however...

@matthew-humphrey or rotate the filament sensor but hard to wire then :grin:

Pathogenius, 2018-10-26:
I've upgraded just the Z/X holders, and I must say this design is brilliant. I now have my X tension set perfectly. I love the visual instruction images, I hate reading endless steps and do prefer to figure stuff out via deduction. Beautiful!

QUESTION: What are those X-idler rod end screws for, maybe to push the rods out of the holder part. This would be one text note for the instructions that might be valuable. Anyways I just screwed them in all the way.

Now I wonder if there might be a Y-axis adjuster similar to this at some point? I managed to get my X into the sweet spot (255-270), and sure would love to get the Y the same, with such ease of adjusting that this solution provides.

gregsaun, 2018-11-18:
@Pathogenius I am going to explain this better in the final assembly guide. If you are on the FB group or Discord server "The 602 Wasteland" then you should find the article explaining it ;-)

Y axis adjustment is planed ;-)

trevjonez, 2018-11-27:
No sweat I already have an adjusted part on hand. just wanted to make sure you knew about it and there is a paper trail for the next person to search. trying to measure my belt it is .9mm and ~1.62mm total which leaves .72 ish for the tooth which is pretty close to what I can find on the spec for a gt3 belt. basically all I did is leave the teethe as they were and pushed the top side up .25mm. its a bit tight and I will need to sand at it to get it in. should have done .3 higher on the part I printed.

as I was looking at your carriage file it looked like there might be some tooth spacing issues as it goes around the bends. perhaps you can lock the pitch line as the ref point and then just do a pattern of whatever tooth profile along a path such that the spacing will auto solve? seems like the whole thing is a bit overly complex constraint wise so a single path with a tooth profile might be easier to deal with. I've not done that in fusion but it seems possible. will play with the concept if I can find the time. at least as a learning exercise for myself :)

gregsaun, 2018-11-28:
wow +0.25mm and still a bit tight, that is a really thick belt!

Part of the teeth are made with pattern but Fusion was not solving the angles correctly to me when doing it in a sketch. I tried to do it out of the sketch and was working better but the mirroring I am using to the other side did not work or was too boring to do. But now you give me another idea I can try :-)

trevjonez, 2018-11-28:
just to circle back the .25 adjustment was perfect once I cleaned up the part with a .8mm drill bit for the tooth grooves and an xacto blade on the flat side to remove any ooze drips etc...

closed 2019-04-04 by gregsaun