gregsaun/bear_extruder_and_x_axis #7: PINDA mount is 1.5mm too close to X smooth rods

gregsaun, 2018-05-24:
Because the X carriage is thinner by 1.5mm, the PINDA is closer to X smooth rods by the same dimension.

Panayiotis-git, 2018-05-24:
My measurements are not perfect (design imported to Tinkercad...), but related to the original design, I measure:
- In your design, the hot-end has moved closer to the X-axis rods by 2.2mm
- The PINDA has moved closer to the X-axis rods by 2.15mm

gregsaun, 2018-05-25:
Hi Panayiotis, thanks for helping me on this. Unfortunately, I have something quite different from my measurements :

X-Carriage MK2.5 from Prusa (STL imported) :
[image]

X-Carriage MK2.5 from Bear :
[image]

Pinda mount from Prusa (STL converted by [jzkmath](https://github.com/jzkmath/Original-Prusa-i3/)) :
[image]

Pinda mount from Bear :
[image]

So 1.5mm delta on X carriage, which means the Pinda mount should move by the same dimension

I will publish soon a fix with a better mount that helps for adjustment as well as maintenance and building.

Panayiotis-git, 2018-05-25:
Ok, you are right about the distances. You are talking about differences with MK2.5 and I measure differences with MK3.

But, I think the difference comes from the X-carriage thickness.
MK3: 15mm (12mm from edge to the center of the rod)
MK2.5: 14mm (11mm from edge to the center of the rod)
Bear: 13.5mm (9.5mm from edge to the center of the rod)

So, the 1.5mm you measure is 11mm - 9.5mm.

gregsaun, 2018-06-07:
Fixed in https://github.com/gregsaun/bear_extruder_and_x_axis/commit/f53affef560640b3b3778b92ea220dfee2910335

closed 2018-06-07 by gregsaun