gregsaun/bear_extruder_and_x_axis #8: F360 source file are outdated

Panayiotis-git, 2018-06-11:
Hi Grégoire,

do you plan to push the latest F360 files soon?

gregsaun, 2018-06-14:
Yes sure, sorry for this. I have a shroud that I do not like yet in the source files and to avoid problem I prefer to wait until I get something good. But this shroud takes much more work than necessary. Next week should be good with interesting new features as well!

Panayiotis-git, 2018-06-14:
Nice! I'm building a MK3 printer and I decided to use your x-axis and extruder design. I needed the source files to make same changes to the parts.
Anyway, since I am impatient, I printed your original parts.
The quality of your designs is superb!
All the bolts and nuts fit perfectly. I didn't use a drill, not even to a single hole.

gregsaun, 2018-08-06:
https://github.com/gregsaun/bear_extruder_and_x_axis/commit/dd3854dc0b37040ee7a7afbaa5813d2e0482567c

closed 2018-08-06 by gregsaun