gregsaun/bear_extruder_and_x_axis #9: X carriage back hits z-top

Panayiotis-git, 2018-06-28:
If the extruder is at the left-most or right-most positions and I increase the Z position, the x_carriage_back part hits the z-top parts.

gregsaun, 2018-06-29:
Sorry for my late answer, could you show me a quick and dirty photo where it is touching exactly?

Panayiotis-git, 2018-07-01:
Hi, sorry for the delayed answer, but I was not near the printer.

I have marked the problem area with a red circle. I'll try to take a photo, but it is a difficult shot.
[image]

Panayiotis-git, 2018-07-01:
Ok, I managed to get the photos.

1st and 2nd are for the z tops.
[image]
[image]

3rd is for the filament spool holder. In fact the holder touches the screw and the hits the orange plastic.
[image]

gregsaun, 2018-08-06:
Solved with new version published yesterday, thanks again @Panayiotis-git

closed 2018-08-06 by gregsaun