gregsaun/bear_extruder_and_x_axis #10: Improve stiffness for everyone and accuracy for Igus users with clamped bearing holders

zbrozek, 2018-08-02:
I was manually torquing my extruder a little bit to see what the least-stiff part of the assembly was. Turned out it was the bearing mounts in the X ends.

For Igus users, the bushings themselves aren't very stiff and they depend on the shell into which they are installed for stiffness. For all users, torquing the extruder will open up the bearing bore a little bit and allow the whole X axis to twist a little.

For reference, [these clamps](https://www.thingiverse.com/thing:2964026) seem pretty nice.

gregsaun, 2018-08-02:
Hi Sasha, thank you for the report. This is something I want to explore as well (plus other places where flex is coming). Good point for the Igus bearings. However, I do not recommend to use Igus bearing, specially on Z axis, they will add too much slop and this is not where the noise is coming from.

I have already updated the design of this part recently (but still no clamp yet) as well as print settings. Could you give me material and print settings you used?

zbrozek, 2018-08-02:
Oh, I'm using the Igus linear bearings because it's what I had on hand when my bearings failed. I had some spares on hand from an earlier attempt to quiet my Y axis. Regardless, the clamps are a nice touch to ensure that the shells can't be torqued open even with ball bearings.

I printed the X ends with 4 perimeters and 30% gyroid infill out of eSun ABS+. My next printer overhaul will make all plastic parts out of some flavor of PC.

gregsaun, 2018-09-20:
This is fixed with clamps on x ends (dev branch for now)

closed 2018-09-20 by gregsaun