gregsaun/bear_extruder_and_x_axis #15: Use pancake stepper motor for extruder.

Graloth, 2018-09-20:
This is just a suggestion, but from the [Skelestruder](https://www.thingiverse.com/thing:2845416) I tested, it seems that using a much lighter slimline motor, doesn't give any negative effects in print quality, and can reduce the weight of the body by quite a bit (https://e3d-online.com/nema17-slimline-stepper-motor). From my testing you'd only need to run a single command (M92 E840) to update the printer to work with a higher resolution motor.

Once I can find some time, I'd like to test this out, initially by just using a slimline motor on your extruder body and see how it fairs, might even be able to move the motor even close to the rods since it weighs less and might not need as much supporting material to sit against.

matthew-humphrey, 2018-09-20:
Skelestruder has a reduction due to the pulley (6:1 I think) which gives it a significant mechanical advantage (more torque from same motor). Using the smaller motor on a 1:1 geared system like Prusa i3 and Bear will not give good results. Just look at all the posts on various forums about people having extruder skipping issues. And if you wanted to try it anyway, what prevents you from swapping the motor with the current design?

I do think that a system like Skelestruder is a great idea. In general I would trade speed for more torque on the extruder motor. But that's a separate issue, unless you are also suggesting that Gregoire add the gear reduction.

gregsaun, 2018-09-20:
Hi Graloth, and thanks for suggesting me this. My objective with the current extruder is to propose an easy upgrade to the stock one without changing too much the hardware and so keeping cost very low. As explained by Matthew you will need to use a reduction and this means we are going away of my objective.

However, I really think we can push this type of frame and axis much more. This will come in a different project based on the bear frame. But some configuration need drastic update and the extruder is only a part of the equation. So more to come but this will take a bit of time (as you might know, I like to take my time to design something good).

closed 2018-09-23 by gregsaun