gregsaun/bear_extruder_and_x_axis #20: Possible part cooling design?

trevjonez, 2018-10-04:
Looks like you are slated to do a part cooling shroud. 

Might I suggest this design as an option? https://www.thingiverse.com/thing:3063554

it will keep the fan vertical thus not having an issue with the wear and tear that the r3 angle might induce. And the design seems to be decently tested for performance already.

I loaded it up and compared with the mk3 parts and it looks like shifting the fan and shroud forward about 1.6-2mm gives it adequate clearance around the pinda.(2mm shown in image)

[image]

robrps, 2018-10-04:
That's mine :) I do plan on doing a Bear version once I've assembled the 0.5alpha to see what's what. I'm quite slow thanks to my 7 mouth old daughter though... Mostly done in Fusion if anyone want the file.

gregsaun, 2018-10-04:
Oh nice to have you here! Happy you are interested in my Bear extruder :-) . I would appreciate to have the F360 file to adapt he place for the PINDA, would you prefer to send me this in private? Be sure that if I use your shroud (or a remix of it) I will add appropriate credits and license!

robrps, 2018-10-04:
I've had your Full Bear on my MK3 since RatRigs first sold the kit its a great design 👍 I'll message you on Thingiverse later on today with the file.

gregsaun, 2018-10-04:
Wow impressive Rob. Hope you like it :-) . Got your message thanks!

trevjonez, 2018-10-09:
I used three washers that are ~0.5..0.55mm thick per fan screw to get everything on nice and go figure it looks just like the above image. Been too lazy to draw up a fan shaped shim to print out as I figure even that would hopefully be a temporary solution.

matthew-humphrey, 2018-10-19:
I've been trying this nozzle out as well. I tried it both with no spacer washers, and with 3 spacer washers each on the two print fan mounting screws and the nozzle mounting screws. So far my results have been a bit better than with the stock Prusa R2 nozzle. I also did the water dish test, and I can see it's creating a circular indentation in the water around the nozzle.

matthew-humphrey, 2018-10-19:
I designed a spacer. I'll post a link once I have a chance to print it and test it out.

robrps, 2018-10-19:
If you are interesting trying out a Bear Extruder (v0.5a) specific version, then you can - https://www.dropbox.com/s/poudc3owi7r05y9/Initial%20Bear%20duct.zip?dl=0 It's not finished yet, but it's working as intended.

ufoDziner, 2018-10-20:
@robrps I'm currently using your duct on my 3 Bear extruders with great success.  Super excited to try out the Bear specific version.  I'll see if I can get them printing tomorrow. 

I would really like the F360 files when you have it complete.  Thanks!

trevjonez, 2018-10-20:
I finally made a shim yesterday. it is a quick and dirty copy face from the front of the extruder body. but works. honestly I was considering doing something similar in tpu just so I could see if it quites the fan down.
[bear fan shim.stl.txt](https://github.com/gregsaun/bear_extruder_and_x_axis/files/2498203/bear.fan.shim.stl.txt)
 rename without txt before using

matthew-humphrey, 2018-10-27:
Interesting, here is the shim that I made. 
[Spacer.stl.txt](https://github.com/gregsaun/bear_extruder_and_x_axis/files/2521465/Spacer.stl.txt)

matthew-humphrey, 2018-10-27:
Interesting. This is the spacer that I made. 
[Spacer.stl.txt](https://github.com/gregsaun/bear_extruder_and_x_axis/files/2521468/Spacer.stl.txt)

@robrps - I tried this latest version and it is not even remotely close to fitting on my extruder. @ufoDziner are you using his version from the dropbox share, or the one that was posted to Thingiverse?

Pathogenius, 2018-10-27:
I'll just forward this little known remix of the R3 shroud, which, if you look closely, you can see the air has less of a fight to get through, and also, less air is allowed out the first exit holes, and is redistributed deeper.

https://www.thingiverse.com/thing:3078174

matthew-humphrey, 2018-10-27:
I don't like the R3 fan position. With those cheap fans, it's going to cause premature bearing wear and eventual fan failure. What I like about @robrps design is that it works great with the original fan position (assuming  you use the spacer, or some washers). I'd really like to see an update of it that doesn't require the spacer.

matthew-humphrey, 2018-10-27:
Here's a version with the Bear logo cutout to save a little plastic (good idea, @trevjonez )
[Spacer.zip](https://github.com/gregsaun/bear_extruder_and_x_axis/files/2521477/Spacer.zip)

robrps, 2018-10-27:
Yeah, my bear duct is a little too tight at the moment - it doesn’t require screws to hold it in place! It also blows a little too much on the block, so without a sock it can cause issues. I’m starting that design again but parametrically so adjustments can be made using parameters. Just getting to grips with Fusion360’s way of doing things. I’m more used to NX but I want others to be able to make changes.

matthew-humphrey, 2018-10-27:
@robrps, It looked like the issue was the mount hole was lowered too much. I'm using your previous design with my spacer now and it's working well. BTW, @gregsaun has been making some changes to the extruder in the dev branch, mostly to improve hotend cooling, so you may want to check to make sure there aren't any fitment issues with the new design.

ufoDziner, 2018-10-27:
@matthew-humphrey I'm using the dropbox version.  It is very close tolerance as @robrps stated, but seems to be working okay.  I mount the shroud to the extruder first, then add the fan after.  Make fitting it much easier.

ufoDziner, 2018-10-28:
@matthew-humphrey Just in case you didn't notice this; part of this shroud goes on the inside of the extruder cover.  It's not like other shrouds.  The inner flange goes inside, then the screw mounting flange goes on the other side of the wall.

matthew-humphrey, 2018-10-28:
@ufoDziner - Yes, that was exactly the issue. Thanks again.

robrps, 2018-10-29:
I've done an update to my bear cooling duct. If you are feeling brave, you can download from the links below. I've finished it during my lunch break at work (Shhhhhhh!!) so I've not been able to print or test as yet. There are 2 version, one similar to the last, the other has a staggered exits at the side.... might be worse! Print with 3 perimeters and a print width of 0.45. 

https://www.dropbox.com/s/u23mmxfw3rlxaw5/Bear%20Part%20Cooling%20-%20Non-Staggered.stl?dl=0
and
https://www.dropbox.com/s/2a102sx6pk2locn/Bear%20Part%20Cooling%20-%20Staggered.stl?dl=0

If I can get the Fusion exporting working I'll link the file. But it's a distributed design so its not letting me. but the angle, exit width, stagger and wall thickness are all adjustable parameters.

trevjonez, 2018-10-29:
I just got the first bear one on and fit and function has been great. I'll give these a spin. only thing I wonder is if we can do anything more to have the air flow angled more downward towards the part/bed?

matthew-humphrey, 2018-10-29:
@robrps - @gregsaun shared with me a technique to get the Fusion 360 export working when it gives you that error about the distributed design. From Gregoire:
1. Within Fusion 360, click on top of your project (note _project_, not design - this is in the pane on the left hand side of the screen, at the very top). This will open the project in a web browser.
2. Navigate to your model, and then click the Export icon.
3. Choose the F360 format.
4. In a few minutes, F360 will send you an email with a link you can use to download the exported design (in a .f3z file).

robrps, 2018-10-29:
Thanks @matthew-humphrey although I've now realised that by bringing in the whole of Greg's Bear Extruder and X-Axis means that it now is going to export it all as well!...Oh well, next iteration I'll just use the Extruder Cover file or create non associative reference/skeleton geometry. Link to the file below.

The above 2 STLs are a little too close to the PINDA, plus the little tab cut out is needed, but currently missing (along with the Bear text and strengthening ribs.)

https://www.dropbox.com/s/p5dttqnfxyzx4j9/Alternative%20Part%20Cooling%20Duct2.f3z?dl=0

matthew-humphrey, 2018-10-30:
@robrps, Gregoire has been making some changes to the Pinda position recently. I think in the last iteration he moved it back, but you might want to check before you make any more updates.

BTW, you guys have any tips on printing this in ABS? I bet some bad overhang drooping in the openings where the air comes out.

robrps, 2018-10-30:
@matthew-humphrey The probe position is not an issue, because it's a parametric design it's trivial to adjust. Normally I print these parts with the custom supports, but they are usual the last thing I create. The exits are now straight and not curved, so you'll need to adjust your bridge settings or use supports from the build plate, or custom supports with the new Slic3rPE.

ufoDziner, 2018-10-30:
@robrps So, would you consider this ready to test?  I can print a few more of them for testing.  Thanks!

robrps, 2018-10-31:
So the staggered version made no difference that I could see from the non staggered version. So it's probably only useful to clear any dramatic Pinda position design changes in future versions of the Bear Extruder.

I made some tweaks to the model.
https://a360.co/2OX0eKC Fusion360 shared link so you can download the F3Z file. It annoyingly still wants to add all of the CAD for the extruder and X-Axis.
https://www.dropbox.com/s/rn3jdqvdpxnorqo/Bear%20Duct%20V3.zip?dl=0 for the STL (including a no text version for @trevjonez !)

@ufoDziner This should be the last design change until the next version of the extruder gets released. Unless something is broken/not working of course!

@gregsaun please feel free to use any or none of this model, the dependences on your parts means it's probably easier if it were put into your Fusion360 design if you wanted to use it for your Bear Extruder.

trevjonez, 2018-10-31:
@robrps setting up the latest for print now 👍 such good community here.

ufoDziner, 2018-11-01:
Wow!  I thought the old version was a tight fit.  There is no way that any air is leaking around where the fan goes down into the shroud.  My only real issue with the old one was that it came very close to touching the parts.  We'll see how this one is.  Thanks again @robrps!

trevjonez, 2018-11-01:
Not as straight as I wanted but to not too bad for a first attempt at hand placing.
[image]

ufoDziner, 2018-11-01:
Mine are done as well!  Printed in ABS+.

[image]
[image]
[image]

matthew-humphrey, 2018-11-05:
@robrps - FYI, your design does not fit the latest iteration of the Bear extruder in the Dev branch. I'm attaching a screenshot of the Fusion 360 interference check between the cooling duct and the extruder cover.
[image]

robrps, 2018-11-06:
That looks like an easy fix, but I'll wait until the extruder cover design is fixed before adjusting. I don't have a lot of free time at the moment.

matthew-humphrey, 2018-11-10:
I did a little surgery on the part with an exacto knife, and got it to fit. However, this iteration of the nozzle/extruder is not performing as well for me as even the Prusa B2 nozzle. The early iteration that required the spacer was doing quite well, but not this one.

ufoDziner, 2018-11-19:
@robrps Looks like the the extruder is done for now.  Any chance that you could have a look at the interference?  Thanks!

robrps, 2018-11-21:
I will do, but possibly not a for a week or so. I'm both lacking in time and filament to print off the new extruder at the moment.

robrps, 2018-11-21:
@ufoDziner Untested and unprinted, adjusted for the the 0.6 Bear Extruder. @matthew-humphrey I've angle down the exits - can you let me know if that helps, if not I'll put them back (assuming you've printed the 0.6 bear)

https://www.dropbox.com/s/bmjvias6fifalkm/RPS_Bear%20V0.6%20Extruder%20Part%20Cooling__BETA.zip?dl=0

ufoDziner, 2018-11-21:
Thanks @robrps!  I'm printing it now.  Hoping to have the 0.6 parts installed tonight/tomorrow.  I'll report back.

matthew-humphrey, 2018-11-21:
Thanks, @robrps - I am working on the 0.6 upgrade as well, and will give this a try and report back.

ufoDziner, 2018-11-21:
Fits like a champ!!

[image]

matthew-humphrey, 2018-11-21:
Nice! Tried a print with it?

ufoDziner, 2018-11-22:
Yep.  Works well.  I don't see any difference, but I don't have a "trained eye" either.

Rorrim77, 2018-11-22:
@robrps I was happy to find your shroud as the Prusa I3 MK2.5 shroud does not fit on the Bear Extruder, so needed a new one) Printed and using your part cooling shroud! It was printed in PETG but have some problems of the fins being very weak. One fin broke while trying to install the shroud (the fan sits very tight on it). Don't know if it will effect the cooling but will print another one.  Question though: There is another file with the shroud called support, how is that one used?

robrps, 2018-11-22:
@Rorrim77 You might not be printing hot enough, or are using too much fan for PETG. The ones I've printed in PETG take a reasonable amount of force before breaking.

The support file is not strictly needed, but in Slic3r, load the duct stl as normal, double click on the part in the build tray, then click on load part in the window that pops up and bring in the support stl.

Rorrim77, 2018-11-22:
@robrps It might be too much cooling as I am already printing it at max temp (250 degrees) I will try with less or no cooling hopefully the layer adhesion will be better that way.  Thank you for explaining the support bit. I don't think I really need the support as the print came out perfectly with no supports, but I am going to print one with the support just as an experiment 👍 Keep up the good work.

trevjonez, 2018-11-26:
@robrps hey have you ever tried to design the shroud such that it mounts to the fan rather than the extruder? I have been running a bit longer nozzle than what the e3d profile is and usually end up fighting alignment anyway since the shroud is typically only attached by one screw. was thinking it would be super nice to have the shroud on the fan . then mount the fan to the extruder via an adjustable rail or similar so that it could be easily removed or adjusted.

WarrenSchultz, 2018-12-04:
@robrps I want to make sure I'm not having a bit of end-user error. :) Is the F3Z file linked above the older version, while the beta isn't posted in Fusion format yet?

robrps, 2018-12-04:
@WarrenSchultz correct, the F3Z isn't up-to-date yet. I've yet to print the 0.6 Bear extruder so I can't test anything as yet.

@trevjonez not really, the non-bear version has an arm to mount to the fan, but it's not very aesthetically pleasing!

robrps, 2018-12-05:
Can someone else confirm this for me please - I've been testing some part cooling ducts and overhang print settings and I've found some settings that seems to work very well for overhangs (on my print at least!).

https://drive.google.com/open?id=1AhizWgEqOUzhjJ1RIY_oeQc6YYZYu3Ll

If you have some spare time, can you print the STL in the attached using Slic3rPE with the your standard settings, then try again using the settings from the config.ini. I've adjusted the print line widths and bridging speed. The print line widths are based on the suggestions from here https://dyzedesign.com/3d-printing-speed-calculator/  Way wider than I usually run... 

[image]

ufoDziner, 2018-12-05:
@robrps I can give it a shot.  What type of filament are you using for the test?

robrps, 2018-12-05:
That print was Extrudr PLA NX2.

ufoDziner, 2018-12-06:
@robrps 
The only settings I changed were the filament temps and added my pressure advance settings for the corners. 

[image]
[image]
[image]
[image]
[image]

gregsaun, 2018-12-06:
Hi @robrps sorry not active here but I am reading all the comments, awesome work!

Be careful with the calculation you linked, that is for DyzeDesign hotend and they are different than E3D, they are extruding much wider perimeter with a smaller nozzle.

However I also found improvement by increasing speed of bridging. I am at 30mm/s, what about you?

robrps, 2018-12-06:
@gregsaun This is true, however the E3D 0.4mm nozzles have a 1mm diameter contact surface so it should be fine. I'm bridging at 10mm/s - two schools of thought: print slower to get as much cooling done as possible; or print fast to get the hotend as far away as possible to prevent excess heat in the lower layers from causing warp. Still trying to work out why printing wider is helping. Perimeters at 0.6mm, external perimeters at 0.5mm are the ones from my photo.

@ufoDziner your number 1 print looks way better than mine, but number 2 looks better.

gregsaun, 2018-12-06:
@robrps that is interesting. I don't understand why it is so wide on DyzeDesign hotend as well. If you find some information I am interested :-)

gregsaun, 2018-12-06:
@robrps I am wondering if the middle hole is really necessary, have you tried without?

robrps, 2018-12-06:
@gregsaun I haven't, but I can. I'm off work next week, so I can test then.

gregsaun, 2018-12-06:
@robrps What I suspect is that the middle hole is pushing the air on the back instead of "using" the air under, on the part. Hard to explain, I hope you understand what I mean. In this case it might interesting to open more the lateral holes. That would be nice if you can test this, no rush and take your time ;-)

WarrenSchultz, 2018-12-06:
@robrps I haven't gotten a chance to print this yet, but your split-airflow design is something I'd been wanting to try myself when I got a chance. :)

A few thoughts on the thread: 
I suspect the wider lines are setting faster due to more surface area, with less "core" molten filament to cool. Basically a ribbon vs. a noodle. ;) (Unless you're increasing your volume to match the width, but I didn't get that impression.
Middle cooling hole: I'd like to see the performance with the center hole reduced in size, and/or adding an extra bend (up-down) to increase the distance that air has to travel before leaving the nozzle, to put more pressure behind the side nozzles, but leaving some cooling for straight-on. (I'm still working on my BMG-Bear extruder design, so I can't try it myself yet.

[207 more replies]

closed 2019-06-15 by gregsaun