gregsaun/bear_extruder_and_x_axis #25: Bondtech tightner (single) misaligned

TurboMech, 2018-12-29:
@gregsaun I've been running this setup for a while before the public release and now the bondtech gears are slipping on the filament.  I believe this is due to the hole (for the tightener) being placed over the gear's meshing teeth and not closer to the teeth for the filament.  This is causing the filament to extrude sometime and not extruder other times.  

Although this could be due to body parts warping when printing at high temperatures within my enclosure.

TurboMech, 2018-12-29:
@gregsaun I'm not quite where the issue lies or if you fixed it in the latest revision.  But it seems if the idler door warps even slightly then it will not allow the bontech gears to be close enough to grab filament.  I need some more time to work on this, but I think if the tensioning screw was aligned where the filament teeth are on the bondtech gears I think it may work better and allow for more or less tension on the filament

gregsaun, 2019-03-13:
@TurboMech  sorry for my late answer. It is possible that the Bondtech teeth (big ones) are not meshed together correctly and so the door does not close well. Is it what you are talking about?

closed 2019-04-04 by gregsaun