gregsaun/bear_extruder_and_x_axis #27: [Quick assembly] Z nut missing from part list ?

anthonyda, 2019-01-01:
Hello,
When looking at the instruction/manual, there doesn't seems to be any references to the black Z nut (where the threaded rods goes) that we can see here : https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/manual/quick_assembly.md

Can you please please edit the part list ( here : https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/manual/bom.md ) with the reference to this part ? 

TY

photog0411, 2019-01-03:
It's the ones that come with the printer

anthonyda, 2019-01-03:
Sure but I've got a prusa clone and would like to adapt it to my printer.
So I'm looking for the part reference in order to buy it.

middelink, 2019-02-03:
I guess you want something like this:
https://www.aliexpress.com/item/2pcs-Original-Prusa-i3-MK2-MK2S-MK3-3d-printer-Z-axis-Trapezoid-Motor-Screw-Nuts-For/32860620795.html

nicpottier, 2019-06-04:
Anybody know how these compare to say standard brass ones in 2mm pitch, such as these?
  https://www.aliexpress.com/item/T8-Trapezoidal-Nuts-8mm-Part-Copper-Screws-For-Stepper-Motor-Lead-Screw-Diameter-10mm-Pitch-2mm/1000006527658.html?spm=a2g0o.productlist.0.0.5c955f0b6RwGfq&algo_pvid=579e6ea7-ea7c-4e17-8d71-02795be9bf5e&algo_expid=579e6ea7-ea7c-4e17-8d71-02795be9bf5e-24&btsid=03f8276a-d8f9-404e-acf5-06110dac4432&ws_ab_test=searchweb0_0,searchweb201602_2,searchweb201603_52

nicpottier, 2019-06-04:
Or shorter: https://www.aliexpress.com/item/Geeetech-2pcs-lot-3D-Printer-Parts-M8-Copper-Nut-for-Diameter-8mm-Pitch-2mm-Z-Axis/32821734240.html?spm=a2g0o.productlist.0.0.5c955f0b6RwGfq&algo_pvid=579e6ea7-ea7c-4e17-8d71-02795be9bf5e&algo_expid=579e6ea7-ea7c-4e17-8d71-02795be9bf5e-39&btsid=03f8276a-d8f9-404e-acf5-06110dac4432&ws_ab_test=searchweb0_0,searchweb201602_2,searchweb201603_52

gregsaun, 2019-06-15:
All datasheets will be available with the next release coming very soon.

In the mean time you can find them in the bom on the dev branch:
https://github.com/gregsaun/bear_extruder_and_x_axis/blob/dev/manual/bom.md

closed 2019-06-15 by gregsaun