gregsaun/bear_extruder_and_x_axis #30: Alternative to Sunon MF40100V1-1000U-G99

timwj2016, 2019-01-17:
The Sunon MF40100V1-1000U-G99 is recommended as an alternative to the Noctua fan however the Sunon seems very difficult to source.  Is there an alternative to the Sunon that has similar specs?

Thanks

Tim

timwj2016, 2019-01-17:
I am in the US and everyone, including DigiKey, seems to have the A99 but not the G99.  I ordered the Delta and will give it a try.
Thanks

mmarbut, 2019-01-18:
I got mine here out of stock for now though.
https://all3dmakers.com/products/sunon-mk2-5-mk3-5v-hotend-cooling-fan?variant=17134119780467

j00sef, 2019-01-30:
All3D Makers have the fan where you can splice it to your existing cable if you can't wait for the pre-assembled ready to go one.
[https://all3dmakers.com/products/sunon-mf40100v1-1000u-g99-hotend-cooling-fan](url)

matthew-humphrey, 2019-01-31:
I tested the Delta fan that I linked above. I have a temperature probe and PTFE tube assembly that I can use to sense the temperature in the critical region of the heatbreak. I ran some tests with hotend at 260C and extruder just above an 80C heatbed. With the Sunon V1, the temperature in this region of the heatbreak was 35.8C. With the Delta it was 36.5. With the Noctua, it was 40.9.

So although the Delta is not as good, it is very close. I found the noise levels similar to the Sunon V1.

And yes, All3d has the Sunon V1 now, so you can just go with that.

gregsaun, 2019-04-04:
I have added the Delta to the dev branch now. Thanks everyone for helping me on this topic and thank you again @matthew-humphrey for having find the Delta!

closed 2019-04-04 by gregsaun

rrauenza, 2019-04-24:
I happened to be researching the different specs of the Noctua vs Sunon in order to upgrade my stock mk3’s extruder fan and came across this issue - so I thought I’d also document them here for others.

What I found surprising is the difference in static pressure between the Noctua and the Sunon - 1.78mmH2O vs 4.83mmH2O (0.9inchesH2O) - almost a 3x difference.  Volume is 8.2m^3/hour vs 13.6 (8 CFM). 

Of course it is also has a higher rpm and is louder ... but reportedly helps with jamming.

https://noctua.at/en/nf-a4x10-5v/specification
http://www.sunon.com/uFiles/file/03_products/08-catalog%20download/Sunon%20DC%20Brushless%20Fan%20&%20Blower_(240-E).pdf

[image]
[image]

austinvojta, 2019-05-22:
How about this one from CUI? Looks like an exact match to the Sunon V1. Same CFM, SP, and noise level, all while also being 5V with tachometer signal:

CFM-4010V-070-273-20

https://www.cui.com/product/resource/cfm-40v.pdf

adudek16, 2020-05-17:
Octopart.com seems to be a good source to find who has stock.

tcurdt, 2020-07-13:
Just to clarify: The A99 is not an acceptable alternative to the G99?