gregsaun/bear_extruder_and_x_axis #32: Allow motor removal without tearing down extruder

guykuo, 2019-03-18:
I have been experimenting with 0.9 degree motors in x, y and extruder positions and having good success in all three positions. 

https://shop.prusa3d.com/forum/user-mods-octoprint-enclosures-nozzles--f65/stepper-motor-upgrades-to-eliminate-vfa-s-vertical-t28098.html

One suggestion I have is to allow motor removal without needing to tear down the entire extruder. This can be achieved by...

1. Shifting lower left, rear cover bolt laterally.

2. Add access holes through mounting plate to reach motor bolts. These holes would go through the upper bearing mounts and ideally be large enough to fit M3 bolt head.

If these simple changes were made, motor swaps could be performed by removing the upper rear cover, slipping the upper bearings out of the way, and removing the four motor bolts. This would dramatically simplify swapping and servicing of the motor. The present design requires a much more extensive tear down to change motors.

My rough artwork below shows the suggested changes.

[image]

[image]

guykuo, 2019-03-21:
Here are some STL's for the MK3 that implement the requested changes. Sorry for the rough editing. This is my first time using Fusion 360

[Kuo motor service Bear X Carriage Mk3.zip](https://github.com/gregsaun/bear_extruder_and_x_axis/files/2991110/Kuo.motor.service.Bear.X.Carriage.Mk3.zip)

guykuo, 2019-03-24:
Added holes to top rear carriage cover as well. Now you don't even need to remove the cover nor detach the filament sensor cable to undo lower motor bolts. Just reach in with wrench.

While I was at it, took care of another annoyance - dropping those little white washers when idler door is removed. I modified extruder body and idler door so they hinge independent of the motor. Printing the body with the mod requires a support. I have added a Slic3r 3mf file that has all the modified parts and support enforcer already in place. 

Long bolt that used to be axle for idler door may optionally be cut and to engage nylock instead of motor. 

Here I have removed the motor without removing printer from its enclosure. Just reached in from front of enclosure to do all the work. Fans, hot end, PINDA, filament sensor, and rear carriage covers were left in place during motor removal and replacement.

[image]

Here are updated STL's and 3mf file
[Kuo motor service Bear X Carriage Mk3 b3.zip](https://github.com/gregsaun/bear_extruder_and_x_axis/files/3000174/Kuo.motor.service.Bear.X.Carriage.Mk3.b3.zip)

gregsaun, 2019-04-04:
@guykuo sorry for my late answer, I try to take tasks one after the others :-D . Huge thanks to your research on motor, this was something I planned to do since a long time. I could not find time yet to read everything yet but for now your results are matching perfectly my feelings.

I did thought about this and I drop that idea. The carriage back with screws instead of zip ties is having quite a lot of torque and with screws not centered on the bearings you might twist it a little. This will probably not something that will be true for everyone but I prefer not to take the risk as those axis are very sensitive.

However, just thinking now, maybe we could give access to motor screws in diagonal and using a screw driver with a ball at the end. Do you think we have room for that?

Finally, you did a great job on the design too, congrats!

guykuo, 2019-04-04:
I understand your theoretical design concerns regarding bearing alignment, but I doubt they are a real life problem. Your x-carriage and cover are fairly stiff and conform tightly to the bearings. They just don't budge once bolted - even with moderate torque and my offset bolt placement. Your designed parts are definitely stiff enough to do the job well. I don't see any changes in x-axis effects on y-faces in my test prints before and after the bolt offsets.

How about symmetrically offsetting the cover bolts. I didn't try doing that I only had a few days of Fusion 360 experience. Symmetric offset combined with the stiffness and excellent conformity of your parts design should allay bearing deviation concerns. 

Angled approach to the motor bolt heads might work, but it must remain under 20 degrees angle or some ball head wrenches won't engage. Also, the angle will worsen as bolt is backed out.
Another feature of straight through bolt access was useful during my 1:3.5 gearbox attachment testing. I needed to change bolts to slightly longer ones. Angled access hole have made bolt removal impossible without further extruder tear down.

gregsaun, 2019-04-05:
I will see if I can give a try with a slight angle to access the screw, I have few ideas.

Oh I did miss the gearbox thing, did you post it somewhere? That looks very interesting. Also, if you want custom motors and are searching something specific please let me know and I can check with LDO to help you. Cannot promise anything, but we can try.

guykuo, 2019-04-08:
Gearbox is now on Thingiverse

https://www.thingiverse.com/thing:3550304

I'm looking forward to your official updated design with screw access. Mine works nicely with the screws and independent idler door, but something official from you would be nice for others. I am refraining from publishing on Thingiverse as a remix, until you do your thing.

BTW, the gearbox + your extruder is pretty awesome. I have the Bondtech kit also in hand, but this is working very well and I retain fast access for experimentation. Going with the actual Bondtech upgrade, I would lose a lot of experimental flexibility but would gain just a little bit better performance.

gregsaun, 2019-05-28:
Hey sorry again for my last answer. I did some test and will not make the motor removal like you proposed. I could not find another solution that works for me so unfortunately this will not come from me.

Your gearbox is getting pretty famous at the moment, several people are using it, congrats! 😃

guykuo, 2019-06-10:
Yes, the BNBSX extruder has really taken off.

I'm surprised that your testing didn't work out, but based on your concerns regarding bearing stability, I'm adding another screw to the upper cover to maintain even force across both upper bearings.

gregsaun, 2019-06-10:
You made a great work with all your testing and design.

I have recently change the bearing pocket to make it less prone to over-tightening, have a look on the dev branch.

The problem with bearing is that it will work for most people but my community is quite big today and I have to be sure my parts work for everybody. For example, people are using Vesconite bearings, old/new Prusa rods, Misumi rods, all sort of chinsium bearings/rods, more professional bearings like IKO and so on. Some people have reported issues with Vesconite already, they were unable to use the 3 bearings on the X carriage.

Having a bearing that is not tightened with "equal force" might results in several issues like crash detection or premature wear of the rods/bearings. It might also generate artifacts in prints that are not always easy to understand or track.

guykuo, 2019-06-10:
That IS a wide variety of bearings and rods to support. No wonder you are so conservative in that respect. Meanwhile the bunnies just keep pushing buttons. Here is what your carriage has morphed into with five screws for the upper cover. With the new 5 screw configuration, the upper cover once more has uniform clamping force along length of both upper bearings.

[image]

Lower cover had to be reconfigured to become hot end safe for R4 height extruders. The original configuration was destroying my hot end cables during XYZ calibration. Also, the bottom edge was too close to heat block and melting when used with short R4 height extruders.

gregsaun, 2019-06-10:
I appreciate you are pushing boundaries and this is exactly why I do my project in open source!

This is better for sure, but you will might have some different flex in the cover with the hole on the right compare to the those on the left. Sorry, still not compliant to my design "rules", but does not mean I am correct :-D

Are you sure the filament sensor cables are not touching when the extruder is full up?

Have you seen that the MK2.5 version reduce print surface on X axis due to longer end stop arm (and cables centered on the back)?

Yes I know for the hotend cables, they fixed it differently now by moving the nut on the heated bed cable cover. I am not very fan of this because you might shortcut bed connection when using a wrench to tighten the nut between the two + and - screws.

guykuo, 2019-06-10:
Would not have been able to get so far without your files. They are barely recognizable now, but you created a fantastic base. It is amazing that the PINDA mount does not melt even with prolonged printing of polycarb. The only problem is screw gradually getting loose on PINDA clamp after a few hundred hours. I just changed PINDA mount to use Nylock instead of M3S to combat that issue.

I noticed the stop arm seemed a few mm longer, but wondered if I had inadvertently shortened mine. I'm not becoming feeble minded.

That rear cable routing also thickens the extruder assembly. Not great for moment arm.

Filament cables are fine. Their rearmost position still clears frame.

gregsaun, 2019-06-10:
Happy you like it and find it useful.

I kept the grid for that but as you mention it is still not totally safe. You can also add a dab of Loctite or reduce the hole diameter to have plastic art acting like a nylock. Pinda is also badly placed because it is very sensitive to temperature, they should really move it somewhere else.

I am really not a big fan of this smaller extruder height. Which is why I stick to older design for now.

closed 2019-06-15 by gregsaun