gregsaun/bear_extruder_and_x_axis #36: X-carriage rod holes too tight

haarp, 2019-06-29:
Hi,

I've printed the X-ends on an i3 MK3 out of PETG with the recommended settings: 0.20mm, 5 top/bottom, 4 perimeters. I only deviated by going with 40% infill instead of 20%. I am confident that my MK3 is as precise and well-tuned as it gets.

Unfortunately I'm having trouble inserting the smooth rods (Igus SWM-08). I'm leaning on the X-ends with all my weight and the rods still won't go all the way. I actually ended up having to hammer on the ends with a mallet. They scrape up a bit of material on their way. Of course this does not improve precision... And I can forget about the "X smooth rods tension" or ever removing the rods again without breaking the ends.

Do you have an idea why this happens? I suspect that the lack of chamfer on the Igus rods might be the problem. Unfortunately I don't have any other rods to test.

gregsaun, 2019-06-30:
Hi Haarp,

The missing chamfer on the rods is probably not helping. Unfortunately, those rods are hardened steel so it will be hard to add a chamfer. You could try with a electrical grinder tool.

My holes have the same size and shape than original Prusa ones. They are a bit on the tight side because Prusa is using a reamer (or drill bit) to make the diameter more precise. You could try to use a 7.95mm reamer on the holes to make them more suitable to your rods.

Note that their is a small pocket at the end of the holes for the chips made during insertion of the rods.

trevjonez, 2019-06-30:
I always use an 8mm bit to ream them by hand, I do it by hand so that I don't accidentally over drill or crack the part. Most drill bits i've measured are slightly under the size they say so it should still come out a snug fit after using an 8mm bit.

vertigo235, 2019-06-30:
I always use the rod and chuck it into my drill and drill the rod into the ends at high speed. The heat generated from friction opens the pocket up just right.

haarp, 2019-07-01:
> Prusa is using a reamer (or drill bit) to make the diameter more precise.

Ah, that makes sense! I did not know that.

In other news, I've taken the plunge and tore apart my current Mk3 to assemble the Bear X axis. Tunrs out the Prusa rods go in much much better than the Igus ones. Still not a perfect fit in all holes, but at least I don't have to grab a mallet :)

They also slide better on the bearings. I guess the surface material and chamfer does make a ton of a difference. Guess I'll be sticking with the Prusa rods.

Thanks for your replies!

closed 2019-07-01 by haarp