gregsaun/bear_extruder_and_x_axis #38: M3x16 Button Head screws missing from BOM for Sunon Fan

harlanaubuchon, 2019-07-05:
Four M3x16 Button head screws are needed to complete the Sunon Fan assembly.

BOM:
https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/manual/bom.md
  
Assembly Guide:
https://guides.bear-lab.com/Guide/3.+Extruder/13?lang=en#s133

gregsaun, 2019-07-05:
It is explained under optional part -> hotend fan but I'll try to make that cleaner.
https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/optional_parts/hotend_fan/

harlanaubuchon, 2019-07-06:
Maybe I'm an outlier but, I ordered all new hardware from the BOM and a new hotend beforehand so I could assemble the vast majority of the upgrade without tearing down my printer.

I don't feel like this project should have to keep track of all the hardware for the optional upgrades but, this one in particular needs screws most people won't have and is not really optional for performance reasons.

I'm happy to contribute some markdown if you'd like a pull request.

gregsaun, 2019-07-06:
I made it clearer in both Bom and optional parts:
https://github.com/gregsaun/bear_extruder_and_x_axis/commit/62619b686c6ef510dce2515593c5fb3f0e496ed6

Thanks for reporting it.

closed 2019-07-06 by gregsaun