gregsaun/bear_extruder_and_x_axis #40: Parameterize trapezoidal nut size?

nicpottier, 2019-08-09:
Howdy howdy. I'm building a bear from scratch and using 8mm lead screws with a 2mm lead. I have two different kinds of trapezoidal nuts (one brass, one POM) and both have nuts 15mm apart instead of the 19mm that the stock Prusa nuts have.

No problem, I can change that in the fusion 360 models (yay!) and I can edit the literals in the sketch there to make it work, but what would you think about these being User Parameters?

Seems we could accomplish this with a single new parameter, the distance between the bolt holes in the nut. Everything else could be based on that.

Main benefit here is that as you continue to evolve then those of us using slightly different things can rebuild our STLs by just changing one parameter.

Happy to make these changes and submit a PR (does that work for fusion 360 files?) if you agree.

nicpottier, 2019-08-09:
So playing with this on my own looks like we would need one more variable as the hole size for the lead screw is actually a bit larger than just `trap_nut_hole_distance - m3_nut`

The above with 19 mm for `trap_nut_hole_distance` (stock prusa) has us come out to 12.9 instead of 13.5 per stock.

[image]

Is this slightly larger hole for reasons of being on even perimeters or something? Just wondering what the right way to model that would be.

nicpottier, 2019-08-09:
Hrmm.. devil is in the details I see, a lot more is dependent on those values than just that, all sorts of errors! Will keep playing with this but also investigate printing a 2mm lead nut with the same dimensions! :)

gregsaun, 2019-12-16:
Hello, sorry for my late answer, thank you for your interesting issue, I agree that it would be great. Have you found a solution for this?

nicpottier, 2019-12-16:
For the time being I ended up just using an adapter. Got a bit stalled on this build so haven't seen how it prints yet but will let you know when I get there!

ChipTechno, 2020-04-16:
I made a simple adapter for 22mm->25mm nuts. With this we can use the standard 22mm nut size on the X part https://github.com/ChipTechno/Bear-upgrade-snakeoil-mod/tree/master/Lead-screw-nut-adapter

trevjonez, 2020-04-16:
I have been running printed nuts on my machine for about a year. It is quite difficult to get good results. 

I am going to give the adapter a try as I have just recently failed to print nuts in pom. 

Currently my nuts are in essentium PA and while functional it has more friction than I want, thus need replacing far more frequently.

trevjonez, 2020-04-18:
I've given the adapters a go and love how they look and fit, but I don't like how constrained the leadscrew is. I end up having to run prints with the nuts pretty loose to not get banding from Z wobble. This is an issue with the pom nuts and imperfect screws though, not your adapters.

If you want to give printed nuts a try, here are the F360 files. I have only personally used the 2/2 on my builds. 
[T8, 2mm Pitch, 2mm Lead](https://a360.co/2J0gFDd)
[T8, 2mm Pitch, 4mm Lead](https://a360.co/2TucnJ6)
[T8, 2mm Pitch, 8mm Lead](https://a360.co/2TD425n)

The latest nylon nuts honestly feel better on the rods than the POM nuts I used with the adapter. I used a .25mm nozzle and .1 layers so that the internal surface is as close to shape as possible. all extrusion widths set to .26, again for accuracy.

I am going to actually design up some shaft couplers using this same concept, see if I can get rid of the wobble with some intentionally placed compliance.

ChipTechno, 2020-04-19:
@trevjonez thank for the feed back. I will try your design when i get the new small nozzle. How often do you need to replace the printed nut?

trevjonez, 2020-04-19:
6 months or so iirc.

gregsaun, 2020-04-29:
@trevjonez would be nice to have this stored somewhere, like this I can share it in community's optional parts

ChipTechno, 2020-05-27:
I ended up by drill 2 extra holes on the nut using a 3d printed template. 
https://photos.app.goo.gl/iB7JpruGCVNLAU9k9

closed 2021-11-27 by gregsaun