gregsaun/bear_extruder_and_x_axis #41: Compatability for linear guideways

nikhil8333, 2019-11-23:
hi, I am trying to make my own design of printer, and tried to accommodate it on the the linear guideway of hgh15 please help me with, the guideway is mounted over the 3030 aluminum extrusion (X- axis rod)) with block facing upward and the printer is a core XY design

gregsaun, 2019-12-16:
Hello, I would love to have time to help you on such a cool project but unfortunately it is not possible anymore today.

I am planning to rework in depth the X axis but we have to finish few ongoing project at the moment (like the Y axis linear rail).

nikhil8333, 2019-12-16:
Well that's sad to hear, but now I am trying with some bull gear model inspiration and will put it on GitHub soon, love to work alongside with you, ping when you get time

gregsaun, 2019-12-17:
What is bull gear model?

closed 2020-03-10 by gregsaun