gregsaun/bear_extruder_and_x_axis #42: Filament sensor is always triggered

Sjoerrdd, 2019-12-12:
Hi,
My filament sensor is always triggered, only if I put it out of the extruder it returns 'OPEN' (M119 command).

gregsaun, 2019-12-12:
May I ask you to share a pic with sensor close and sensor open? This might save some words ;-)

Which printer do you run? MK2.5S or MK3S?

Sjoerrdd, 2019-12-13:
I'm using the MK3S extruder, I'll send a picture when I'm home. (± 5 hours from now)

Sjoerrdd, 2019-12-13:
[image]
[image]

gregsaun, 2019-12-15:
Thank you for your image. I see several issues on your extruder. Could you give me more precision about those points?

- Blue arrow: the carriage is not aligned with extruder body. This could bend the connector and so the sensor PCB making the reading difficult.
- Green arrow: sounds like there is a crack or opening here, make sure not light enters here
- Yellow arrow: not linked to the filament sensor but you are using a pancake stepper which will have not enough torque and loose steps.
[image]

Sjoerrdd, 2019-12-15:
Thanks,
* That's right, I solved that by printing a new carriage, thanks :)
* It isn't a crack, it's a failed layer (I don't have enough PETG to reprint it), when my new roll arrives, I'll fix it
* Pancake stepper motor is temporarily, the new one will arrive later this week. (the original broke :( )

gregsaun, 2019-12-15:
When no filament is inserted you have the opening here that looks too big. Could you find why?
[image]

gregsaun, 2019-12-15:
Also could you try the filament sensor out of the body?

Sjoerrdd, 2019-12-15:
It worked, there were some "pustules" on the back of the lever, so it was pushed too far forward.

Thanks for helping :)

closed 2019-12-15 by Sjoerrdd

gregsaun, 2019-12-15:
Great!