gregsaun/bear_extruder_and_x_axis #44: request x_end_idler scad

CsepregiArtur, 2020-01-19:
Hello

can you please share the x_end_idler scad?

I made an upgrade from an anet A8 but my steel rods are longer, I didn't take that in count that, now I want to test out the printer, but the rods are too long :(.
I would need the scad to modify in that part to be able to fit the longer rods.
this would be only a temporary solution, after that, I want to cut the rods, but still, now I'm not sure its oke all the setup.

gregsaun, 2020-01-23:
I have no SCAD files, everything is designed under Fusion 360. You can find the Fusion 360 source file here with the whole assembly :
https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/src/bear_extruder_x_axis.f3z

gregsaun, 2020-01-23:
Oops closed too quickly :)

closed 2020-03-10 by gregsaun