gregsaun/bear_extruder_and_x_axis #46: Which material to use?

St0RM53, 2020-02-13:
Regarding printing nozzle_fan_duct i have the following materials:
- Prusament PLA/PETG/ASA
- EasyABS
- CPE
- ColorFabb XT-CF20

Which one you recommend using?

Kind regards

trevjonez, 2020-02-14:
The stock prusa fan nozzle is ASA. I would expect the ASA, ABS, CPE HG100 and XT-CF20 to all perform well for the part. The ABS is likely the most cost effective solution.

gregsaun, 2020-02-14:
Yes I would go ASA probably. I am personally using Extrudr GreenTec Pro but it is expensive (but no styrene and biodegradable). If you never tried annealing HTPLA I would avoid trying that on the shroud, need a bit of experience I think.

St0RM53, 2020-02-15:
Thanks i ended up printing it in ASA successfully :)

closed 2020-02-15 by St0RM53