gregsaun/bear_extruder_and_x_axis #47: Hemera fan knocks on the x-motor-end

knickohr, 2020-04-02:
I build up my new Bear 2.1 with the BEARmera. There is maybe a little problem with the dimensions of some parts. The Hemera fan knocks on the x-motor-end while going left to the startup position. So, until there is no endstop switch (MK3s) the fan drives into the left side. On the second picture you see, there is some space left between the x-motor-end and the x- carriage.

Thomas

[image]
[image]
[image]

gregsaun, 2020-04-02:
It is because you are using the old X ends, you need to use the new ones from the dev branch.

knickohr, 2020-04-02:
Really ?  I'm sure I've use the ones from the dev.

knickohr, 2020-04-02:
Ahhh, OK. I see. So I also used the wrong x end idler and the x carriage ?

gregsaun, 2020-04-02:
Not that I see. Sorry, should be easier soon as everything will be available from the github.

knickohr, 2020-04-02:
x carriage back is different too, x carriage and x end idler are OK. Seems I mixed parts. Or I used an earlier dev version ?

gregsaun, 2020-04-02:
No it was available at the same time as the BearMera files. You have to get the X ends, X carriage back and cable guide back from the dev branch.

knickohr, 2020-04-02:
OK, all the parts with date "2 months ago".

gregsaun, 2020-04-02:
This is correct, if you check the X end motor for example you will see that the shape of the arm is different than yours.

knickohr, 2020-04-05:
Fixed ! :-)
[image]

closed 2020-04-05 by knickohr