gregsaun/bear_extruder_and_x_axis #50: BOM for scratch build

sfritz67, 2020-05-12:
Hi,
I'm looking to build a bear Prusa from scratch and had a look at the Additional BOM for scratch build for bear_extruder_and_x_axis.

In the BOM, idler is 2GT 20T, but in assembly guide it is 2GT 16. Which is the right part?

By the way, there are some parts from scratch missing like E3D extruder, bearings, maybe others.

Thank you for your great work.

Fred

gregsaun, 2020-05-12:
Hi, the idler is 2GT 20T but the pulley is 2GT 16T, this might be the confusion.

I'll update the bom, thank you!

sfritz67, 2020-05-12:
If I understand well, idler is bigger than pulley...
Does that mean idler center Z level is different than pulley center Z level to have horizontal belt travel on one side?

gregsaun, 2020-05-12:
Yes, correct. The top portion of the belt, the one attach to the carriage, is flat. The return path is angled but this has no influence.

closed 2021-11-27 by gregsaun