gregsaun/bear_extruder_and_x_axis #59: X-ends with not integrated trapezoidal screw to z-motors

Puebllo, 2021-11-11:
Hi,
I've question, because I'm working on new printer, but instead of using z-motors with integrated trapezoidal screw, I want to use normal nema17 motor with coupler (leftovers from previous projects). I see on images that there's not enough clearence for couplers to fit in "inside" of x-ends (bottom of X-axis will hit couplers as it goes fully down).
Now I'm thinking. May I switch sides of X-ends and rotate them upside down, so the trapezoidal screw will be on the bottom of elements ? Are there any downsides of this solution ? switching sides of logic board and psu are obvious, but is there anything else ?

gregsaun, 2021-11-13:
Hello, I have never tried to install them upside down so I don't really know it that works or not. However, there is this video from Melissa that can help you: https://www.youtube.com/watch?v=N1Hm1EWMIEo

Puebllo, 2021-11-13:
Hey, thank you for response. I saw this video in the past and I couldn't find it. Thanks again. If I will have any question I'll open new issue.

gregsaun, 2021-11-13:
In case of, this video is listed in the optional parts from the community here:
https://guides.bear-lab.com/Wiki/community_optional_parts

closed 2021-11-13 by gregsaun