gregsaun/bear_extruder_and_x_axis #60: Bondtech LGX X-carriage compatibility [MK3S & Mosquito]

gavan1, 2021-11-25:
Hi @gregsaun I have a quick X carriage question, I'de like to practice caution and confirm compatibility. 

I am installing the following 

1. [Bondtech LGX extruder](https://www.bondtech.se/product/lgx-large-gears-extruder/)
2. [Mosquito Magnum Hotend](https://www.bondtech.se/product/mosquito-magnum/)
3. [Bondtech LGX carriage for Mk3S](https://www.bondtech.se/product/lgx-x-carriage-for-mk3s-mosquito/)

Want to confirm these stl files are the correct ones?

https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/optional_parts/bondtech_x_carriage/printed_parts/bear_bondtech_mosquito_mk3s.zip

The above linked bear_bondtech_mosquito_mk3s.zip includes 

- bondtech_mosquito_x_carriage_mk3s.stl
- x_carriage_back.stl

Am I able to to use 3. [Bondtech LGX carriage for MK3S](https://www.bondtech.se/product/lgx-x-carriage-for-mk3s-mosquito/) I purchased  with the stl files (x_end_motor  + idler)?

Thank you and can't wait to build the frame!

gavan1, 2021-11-25:
I have confirmed with Bondtech that it's this firmware.

https://www.bondtech.se/downloads/firmware/prusa/FW3100-Build4993-Bondtech-MK3S-16-EINSy10a-LGX-Mosquito.zip

Despite the MK3S board being the Einsy RAMBo (MK3S/+) and not an EINSy10a.

gregsaun, 2021-11-27:
Hi @gavan1, unfortunately my carriage is made for the previous Bondtech extruder and not for the LGX. I am not planning to support the LGX for now but I am working on a geared extruder.

closed 2025-06-24 by gregsaun