gregsaun/bear_extruder_and_x_axis #63: original extruder body r6 version loose

easonjim, 2022-08-23:
Dear gregsaun,
    I’m printing the original extruder body r6 version and found that the printed circular radiator has a small bayonet, which causes the e3d v6 circular radiator to be too loose. I have calibrated the extruder and there is no problem with the line width. May I ask this question How to solve it?

gregsaun, 2022-08-23:
Hello @easonjim,

The hotend should be loose when it sits only in the extruder body but when you close the extruder cover it shouldn't move anymore. After the step [16](https://guides.bear-lab.com/Guide/3.+Extruder/36?lang=en#s390) the hotend should move a little but after step [17](https://guides.bear-lab.com/Guide/3.+Extruder/36?lang=en#s391) it should be tight.

easonjim, 2022-08-23:
Yes, but I printed the original R6 print model and found that the effect is loose, which is very abnormal;
May I ask what printing needs the society?

gregsaun, 2022-08-23:
Could you please share me a picture of your extruder and the bayonet you have printed?

gregsaun, 2022-08-23:
I see you have opened the [issue on Prusa's Github](https://github.com/prusa3d/Original-Prusa-i3/issues/184). I am not developer at Prusa nor working for them, I just make compatible parts with some improvements. I will close this request as it seems an issue with the Prusa extruder and not my work.

closed 2022-08-23 by gregsaun

easonjim, 2022-08-23:
Ok, thank you very much gregsaun, I will go to the official and wait for the latest feedback.

gregsaun, 2022-08-23:
Try to contact the Prusa support, it might be faster to get an answer :)