gregsaun/bear_extruder_and_x_axis #64: Two different Versions for the x_carriage.stl / bondtech_e3d_x_carriage_mk3s.stl

juckendesAuge, 2022-11-09:
Hello,
I would like to convert X-axis and extruder to Bearmod on my Prusa MK3S+.
Does it work with the + version of MK3S?
Now I am unsure which file to take from the two. Both files are added on 2020-05-11. So unfortunately I can't see which file is newer either.
They are also visibly a little different. In x-carrige.stl, for example, you can still see two hex nut cutouts inside.
The bondtech_e3d_x_carriage_mk3s.stl has 4 holes with recesses for the head, but no recess for hex nuts. The shape also varies a bit.

https://raw.githubusercontent.com/gregsaun/bear_extruder_and_x_axis/master/optional_parts/bondtech_x_carriage/printed_parts/bear_bondtech_e3d_mk3s.zip

bondtech_e3d_x_carriage_mk3s.stl 

sha256sum bondtech_e3d_x_carriage_mk3s.stl 
853fde48b2968c2cbe761dc8a83abadeb01165c71c8c14138ffb7bb69d4f5f1e bondtech_e3d_x_carriage_mk3s.stl

726K 11. May 2020 bondtech_e3d_x_carriage_mk3s.stl

---

https://raw.githubusercontent.com/gregsaun/bear_extruder_and_x_axis/master/printed_parts/bear_extruder_x_axis_mk3s.zip
x_carriage.stl

sha256sum x_carriage.stl 
c4418dcf2ee585574a1f5a4555060b7123b435004d2fb96999e997212a4203c1 x_carriage.stl

701K 11. May 2020 x_carriage.stl

Thank you very much for the help.

Translated with www.DeepL.com/Translator (free version)

gregsaun, 2022-11-09:
Hello @juckendesAuge,

Yes it is compatible with the "+" version :)

The carriage depends which extruder you want to build. The standard Bear extruder that you have to fully print yourself then you will find all information here:
https://guides.bear-lab.com/c/95jghwzvw9l46s3g

It is possible you have the Bondtech extrude which looks like this:
https://www.bondtech.se/product/prusa-i3-mk3s-extruder-upgrade/
In that case it is using a different carriage and you have all information here:
https://guides.bear-lab.com/c/rmf45dv3tqb5uf69

I hope this was helpful :)

juckendesAuge, 2022-11-09:
Hello @gregsaun,

thank you very much for the quick reply. :)

I'm building it with the standard V6 hotend.
I'm concerned with no more scratches in the smooth X-axis rods and easier belt tensioning.

gregsaun, 2022-11-09:
All good with the V6, trusty hotend!

For the scratches make sure to always keep the rods with some grease and clean the dust on the ends regularly. Put some grease inside the bearings before installing them. Another good trick is to print both x_end_idler and x_end_motor together, like if there is a slight variation during the print it will be replicated on both and make them equal.

Enjoy the build :)

juckendesAuge, 2022-11-09:
I clean all Misumi bearings with 99% isopropanol in a glass. After drying, I press lithium-based grease into the bearings.
The Misumi bearings of the X and Y axis I replaced completely with new ones. The four bearings of the Z axis I wanted to clean, grease and continue to use.

Unfortunately I have already printed both x_end_idler and x_end_motor separately, because pet-g pulls otherwise something strings.

gregsaun, 2022-11-09:
Don't leave the bearing too long in IPA as they could rust (balls inside are not stainless steel).

For stringing, if your filament is from good quality and dry you can try to reduce the temperature a little and try to increase retraction speed to 50mm/s. I don't know your wohle setup so take this with a pinch and salt ;)

juckendesAuge, 2022-11-09:
Ok.
Thank you.
Standard MK3s+ with Prusament PET-G.

Should I print the two parts together again, or test first to see if it works with the two I printed separately?

Thank you very much.

gregsaun, 2022-11-09:
If the both prints look good it is not necessary to reprint them 👍

juckendesAuge, 2022-11-10:
Thank you very much. :)

The parts look very good.

The camera can not capture it so well.

[image]
[image]
[image]
[image]
[image]
[image]

juckendesAuge, 2022-11-10:
@gregsaun 

https://guides.bear-lab.com/Wiki/Bear_FAQ#Section_I_am_using_the_Bear_MK3S_extruder_and_I_cannot_pass_XYZ_calibration_How_can_I_get_it_to_calibrate

Quote:
Firstly, you can flash temporarily an older MK3 firmware, such as 3.7.1, and perform the XYZ calibration. Once calibration is passed, flash the MK3S firmware of your choice to access the full functionality of the MK3S.

Does the Bear Extruder work with the latest firmwae from Prusa (firmware 3.11.0)?
Unfortunately I can not flash the 3.7.1, because the Super Pinda is supported only from firmware version 3.9.2.

Thanks a lot

gregsaun, 2022-11-11:
It does but not for the calibration, just use our Bear firmware as recommended.

juckendesAuge, 2022-11-12:
ok.
Thank you very much.
This is also the penultimate firmware version that was modified. Then it is ok.
I use the Experimental feature:
E-cool mode
https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213

I would like to continue using it.

Many greetings

juck.

closed 2025-06-24 by gregsaun